robotology/cer #2: CER depends on iCub repo

barbalberto, 2015-10-19:
Just a question / dicussion:

Shall CER repo depends on iCub's repo?
Right now there are some include files that throws in the dependency like:
```#include <iCub/iKin/iKinFwd.h>```

Anyway, aside from this, the robot will use the iCub devices like the embObj* so the dependency is quite strong right now.

There was an idea to move the icubmod on a separated repo, in that case shall we move also some other reusable library like iKin?

pattacini, 2015-10-19:
Moving `iKin` outside `icub-main` is not an easy job, hence for the time being the answer is ... **`Yes`**

lornat75, 2015-10-19:
The embObj dependency is dynamic right?

For iKin, I think we should take the occasion to start experimenting with YCM for CER and icub-main so that libraries get their own life (or we reduce iCub-main until it becomes a very thin collection of libraries).

@drdanz

pattacini, 2015-10-19:
@lornat75 

I deem this is not the right period to introduce the _noise_ in our eco-system that such a change will bring about, since `iKin` is a widespread dependency for a huge amount of software. Could we wait after the first year demo?

Making `iKin` standalone is also one of my personal goal, so don't take this opinion as an _a priori_ objection. It's just a matter of timing.

In the meantime, experimenting with `YCM` could be perfectly doable with the caveat not to touch the `icub-main` **master**.

drdanz, 2015-10-19:
With "experimenting with YCM" you mean for the icub-main superbuild or as a dependency?

barbalberto, 2015-10-19:
BTW, I updated all and I get the following include error:
/usr/local/src/robot/cer_WS/cer/libraries/cer_kinematics/include/cer_kinematics/utils.h:23:31: fatal error: iCub/iKin/iKinFwd.h: No such file or directory
 #include <iCub/iKin/iKinFwd.h>

pattacini, 2015-10-19:
it's compiling on `Bluey`...
Probably some missing variables in the path.

randaz81, 2015-10-19:
I'm against moving anything out from icub-main UNLESS someone first setups a working superbuild...

barbalberto, 2015-10-19:
The error shows up when the tripodMotionControl is enabled.
Commit 10b03c9 introduced the include and therefore the dependency from iCub. The dependency was handled by the cmake of cer_kinematic library but not updated in the motionControl device.
I'll add this include path in the motionControl as well, but it'll be fancy if this dependency remains in the lib  and not propagated making this icub's include "private". Not a big deal however.

pattacini, 2015-10-19:
Cannot be made "_private_". `cer_kineamtics` needs to expose the `iKin` layer.

pattacini, 2015-10-19:
Why should `motionControl` depend on `iKin` if the dependency is only required by `tripodMotionControl`?

barbalberto, 2015-10-19:
By motionControl I mean tripodMotionControl (I'm lazy)
My idea is that tripodMotionControl should not directly depend on (include by chains of includes) iKin since no iKin object are directly used by tripodMotionControl.
The cer_kinematic library obviously depends on iKin but, by means of tripod.h, it can expose some methods to tripodMotionControl while keeping the dependency (implementation) private. 
In this way the implementation can change without the higher tripodMotionControl to notice. In this case the compilation would not have been affected.

pattacini, 2015-10-19:
I know what you mean @barbalberto, but the library has grown and now contains other stuff for the whole arm that in turns needs to expose `iKin` layer.

Probably the export of the library needs to be tuned up so the user is not forced to explicitly put iCub dependent directives in the cmake files.

closed 2015-10-22 by pattacini