robotology/cer #5: Some problems arising during first tests on the tripod motion control

pattacini, 2015-10-29:
After the preliminary tests we run on the control of the tripod we can report on the following problems we need to address:

- [ ] `yarpmotorgui` needs to be extended to expose fractional slider steps (see https://github.com/robotology/yarp/issues/628).
- [ ] the state port streams out the `pitch` and the `roll` in radians instead of degrees (fixed via https://github.com/robotology/cer/commit/641bb41b826e95be73e42597eaa04b517ef9a41c).
- [ ] the tripod ikin needs to be updated to account for the first roto-translation matrix.
- [ ] the tripod motion control is required to compute proper speed on the fly for each joint so that the joints will attain the desired targets at the same time.
- [ ] we have to make sure that both the FW and the ikin layer will use the same structure's bounds.

These are all minor bugs since we were happy to see the tripod moving correctly in most configurations.

/cc @ale-git @barbalberto @randaz81

ale-git, 2015-10-29:
Fixed low level tilt limit stuck bug by https://github.com/robotology/icub-firmware/commit/1eb755bff1ab7a2972f9bf9cb7fb03e9477b5e09

ale-git, 2015-10-29:
The high level tilt limit has been deacreased to 25 degrees, the same in the low level control. 25 degrees is close to the mechanical limit.

pattacini, 2015-10-29:
@barbalberto could we imagine to try to meet the _on-the-fly speed computation_ requirement by tomorrow so that we can run a further test before the meeting?

barbalberto, 2015-10-30:
I'll start working on it right now

pattacini, 2015-10-30:
Great! :+1:

barbalberto, 2015-10-30:
Commits done!
Let's meet monday morning to re-test everything!!!

@ale-git @pattacini

pattacini, 2015-11-02:
@barbalberto is the `seg-fault` fixed?

barbalberto, 2015-11-03:
On my laptop yes, final word on this to @ale-git

ale-git, 2015-11-03:
I haven't seen seg-faults any more after the fix.

pattacini, 2015-11-03:
:+1: 
Let's tick this as fixed then.

pattacini, 2015-11-03:
Hi @ale-git 
We've identified two new points (I've reported in the first post) we have to go through before considering the deadline met. Could you work on these?

ale-git, 2015-11-03:
The setpoint issue is fixed, the calibration requires a little more time, I'll manage to close it today anyway.

pattacini, 2015-11-03:
Great! Thanks.

pattacini, 2015-11-11:
`yarpmotorgui` is under test (see https://github.com/robotology/yarp/issues/628) but it implements already the fractional step sliders in a dedicated branch, which is waiting to be merged into master.

All the other issues have been addressed successfully, closing...

closed 2015-11-11 by pattacini