robotology/cer #31: Test CER michrophones.

randaz81, 2016-05-03:
Not implemented yet.

barbalberto, 2016-05-03:
Low priority

barbalberto, 2016-05-18:
Spoke with Francesco Diotalevi. The HW configuration of mics is the same as the display. 

The FPGA is in charge of acquiring data from the microphones and made them available to the OS. This  is because later on a beam forming algorithm will (probably) be implemented inside the FPGA.

A custom character driver has been made by Francesco to read the data.

Since it is a character driver, it is not seen by the OS as a sound card, therefore the standard utilities that deals with audio devices cannot be used, neither portaudio.

There are 2 possible ways to go:

1) Create a linux driver for sound card, so that any standard library can work with (like portaudio). More complicated.
2) Create a custom yarp driver reading from character device. Easier, but only yarp SW could use the microphones.

lornat75, 2016-05-18:
I think we can bring this issue at the next meeting and discuss the priorities. I would be better to implement a proper linux driver, if we have time and resources.

randaz81, 2021-06-08:
Addressed in https://github.com/hsp-iit/research-development/issues/73

closed 2021-06-08 by randaz81