robotology/cer #41: Stability of Ubuntu machines

lornat75, 2016-05-03:
Looks like CER machines with Ubuntu have several stability issues. 

@drdanz @pattacini @randaz81

pattacini, 2016-07-12:
Ubuntu isn't really a problem by now.
I'd rather change this issue in **We need to replace the cube with a laptop** :smile: 

cc @randaz81 @mbrunettini @lornat75

pattacini, 2016-07-13:
Actually, we could try to go with only one laptop, preparing the Windows one we have with Ubuntu. I think it could be sufficient.

The **Geomagic** can run on linux; what about the **Oculus** @randaz81?

drdanz, 2016-07-13:
The Oculus, if they didn't update the linux driver, must run on windows

pattacini, 2016-07-13:
If I'm not wrong I've heard @randaz81 saying once that the linux driver of the Oculus is now available.

mbrunettini, 2016-08-03:
A mobile workstation with windows has been ordered

closed 2016-09-25 by pattacini