robotology/cer #46: Message queue not served @ controller rate = 100 Hz

pattacini, 2016-06-24:
We have experienced big problems while controlling reaching with a rate of `100 Hz` (i.e. sample period `Ts = 0.01 s`). What happened was that the internal messages queue got increasing and, as result, the movements got delayed of significant amount of time.

By changing `Ts = 0.1 s`, we improved the situations, even though a rate of `10 Hz` is too low...

cc @randaz81 @barbalberto @ale-git

barbalberto, 2016-06-26:
This was related to issue #45, and thus solved by fixing issue #45.

In general case, we see this problem when a positionMove or setPosition are called on tripodMotionControl at "high" rate. 
As a rule of thumb if this error happen again on CER robot verify that the code is not calling a loop over single joint for tripod device or position command are not sent at too high frequency.

pattacini, 2016-07-12:
I deem we can close it for now.

closed 2016-07-12 by pattacini