robotology/cer #47: Cameras' custom FW do not allow 640x480 resolution

pattacini, 2016-06-24:
Apparently, the new FW we requested for the cameras to change the brightness values do not allow running in 640x480 mode, but only with higher resolutions. Obviously, this is very problematic in terms of band consumption, and we cannot go back at the moment to the previous FW version.

As temporary workaround, @barbalberto is customizing the device driver with the goal of cropping (to switch from 16:9 to 4:3) and then resize down the image (to 640x480) at the robot side.

cc @randaz81 @vtikha

randaz81, 2017-07-16:
@barbalberto can you update (and eventually close if fixed) this issue?

barbalberto, 2017-07-17:
This has been included in the latest PR related to usbCamera.
User can specify **any desired resolution** when starting the device. OpenCV will crop and rescale the source image to best fit the requirement. This obviously increase the workload on the robot side.

closed 2017-07-17 by barbalberto