robotology/cer #51: Test campaign to assess WiFi link quality

pattacini, 2016-07-05:
A thorough test campaign must be carried out to verify the functionality of the WiFi in noisy environments, to then come up with possible solutions to improve the quality of the link.

For example, we should start off by better accommodating the internal antennas.

cc @maggia80 @randaz81 @fiorisi

pattacini, 2016-07-05:
Regarding this, we did experience heavy link drops in the stand and in the arena.
Those environments were particularly noisy in terms of WiFi interference, but these will be the operating conditions in next competitions/scenarios (e.g. open space)

fiorisi, 2016-07-06:
I can take care of the design of a support for the antennas. I will discuss with @maggia80 about the correct spacing.

pattacini, 2016-07-14:
@mbrunettini you can use this issue if you would like to add any comment on the CER WiFi.

pattacini, 2016-11-10:
As quick update, we have been observing quite a significant improvement on the use of WiFi.
In the last two demos we ran the robot without any problem.

However, in both cases, the environment was fairly neat in terms of WiFi interference.

closed 2017-09-15 by pattacini