robotology/cer #58: Head control

pattacini, 2016-07-13:
As of now we cannot control the head, mainly because of mechanical low level problems.
We should thus address this problem at the mechanical stage as much as possible.

This is a note for the design group.

cc @fiorisi @randaz81 @ale-git

pattacini, 2016-09-13:
#### Update 

The encoder has been accommodated on the motor side in order to reduce the impact of backlash.

@ale-git is now coding the modifications to the `FW` required to handle the sensor input.

randaz81, 2016-09-13:
This improves the controller stability, but does not solve the backlash issue, so the positioning accuracy will be poor.

pattacini, 2016-09-14:
With the new `FW` it seems we gained quite a nice improvement in the control.

pattacini, 2016-09-14:
Tested the `gaze controller` on top and it works pretty well and graceful.
I think we can consider this issue solved by now.

Let's not forget to mount a second AEA at the joint level to achieve sufficient precision during navigation.

closed 2016-09-14 by pattacini

pattacini, 2016-09-14:
cc @miggia

miggia, 2016-09-14:
:kissing_heart: