robotology/cer #63: Xtion frames

pattacini, 2016-09-14:
As of now we have one frame attached to the mechanical center of the `Xtion`.

For control purpose, we would need to:

1. Move the frame above called **`depth`** to the actual center of the depth sensor.
2. Introduce a further frame we can call **`depth-rgb`** to be attached to the center of the rgb sensor.

@fiorisi is going to deal with this request (not urgent) next week.

cc @randaz81 @barbalberto @vtikha

pattacini, 2016-09-14:
#### Note
Other frames already available are:
- `left`
- `right`
- `cyclopic`, attached to the middle of `left` and `right`
- `gaze`, attached to the middle of the screen

pattacini, 2016-09-16:
@fiorisi please also consider that Marcello is willing to mount asap the new mechanical support of the Xtion, which will entail to move the two RGB cameras slightly apart (~ `2 mm`).

Therefore, could you take this into account and provide new DH parameters for the cameras' frames as well?

fiorisi, 2016-09-19:
Yes, no problem.

fiorisi, 2016-09-19:
regarding the Xtion, do you want 2 different <sensor> tags (one for the depth sensor and one for the rgb sensor) or do you have a solution to embed the camera inside of the depth (or vice-versa)?

pattacini, 2016-09-19:
I'd go for **`depth`** and **`depth_rgb`**, as reported above.

barbalberto, 2016-09-19:
The most flexible solution appears to have 2 different sensors in the urdf/sdf, a rgb and a depth camera.
In this case, playing with config file, we could have the `realistic` robot configuration (rgb and depth camera NOT colocated) or the `ideal` configuration where the 2 sensor are colocated.
(or both)

pattacini, 2016-09-21:
With @fiorisi we agreed to keep in the model also the fake frame attached to the center of `Xtion` device for control purpose. It will be termed **`depth_center`** (actually it's not cyclopic)

fiorisi, 2016-09-22:
Testing the following frames: `depth_center`, `depth` and `depth_rgb`.
The test are performed by computing the transform between the `base_link` and the `frame to be tested` using our forward kinematics library and our CAD model.
The position of the joints that belong to the kinematic chain (between  `base_link` and the `frame to be tested`) is defined by the vector **q**. The CAD model is updated accordingly. 

## frame=depth_center
**q=( 0.00000  0.00000  0.00000  0.00000  0.00000  0.00000)**

_IPOPT_
H=-0.00000  0.00000  1.00000  0.02156
-1.00000  0.00000 -0.00000  0.00000
-0.00000 -1.00000  0.00000  1.07240
 0.00000  0.00000  0.00000  1.00000

_CREO_
[image]
---
**q=( 0.01500  0.00000  0.00000  20.00000  15.00000  0.00000)**

_IPOPT_
H= 0.33993 -0.34839  0.87354  0.13368
-0.93969 -0.08852  0.33037  0.00941
-0.03777 -0.93316 -0.35747  1.04557
 0.00000  0.00000  0.00000  1.00000

_CREO_
[image]
___
## frame=depth
**q=( 0.00000  0.00000  0.00000  0.00000  0.00000  0.00000)**

_IPOPT_
H=-0.00000  0.00000  1.00000  0.02156
-1.00000  0.00000 -0.00000  0.04449
-0.00000 -1.00000  0.00000  1.07240
 0.00000  0.00000  0.00000  1.00000

_CREO_
[image]
___
## frame=depth_rgb
**q=( 0.00000  0.00000  0.00000  0.00000  0.00000  0.00000)**

_IPOPT_
H=-0.00000  0.00000  1.00000  0.02156
-1.00000  0.00000 -0.00000  0.01849
-0.00000 -1.00000  0.00000  1.07240
 0.00000  0.00000  0.00000  1.00000

_CREO_
[image]

fiorisi, 2016-09-23:
Testing the following frames: `left`, `right`, `cyclopic` and `gaze`.

## frame=left
**q=( 0.00000  0.00000  0.00000  0.00000  0.00000  0.00000)**

_IPOPT_
H=-0.00000  0.00000  1.00000  0.01385
-1.00000  0.00000 -0.00000  0.03351
-0.00000 -1.00000  0.00000  0.96090
 0.00000  0.00000  0.00000  1.00000

_CREO_
[image]
___
## frame=right
**q=( 0.00000  0.00000  0.00000  0.00000  0.00000  0.00000)**

_IPOPT_
H=-0.00000  0.00000  1.00000  0.01385
-1.00000  0.00000 -0.00000 -0.03351
-0.00000 -1.00000  0.00000  0.96090
 0.00000  0.00000  0.00000  1.00000

_CREO_
[image]
___
## frame=cyclopic
**q=( 0.00000  0.00000  0.00000  0.00000  0.00000  0.00000)**

_IPOPT_
H=-0.00000  0.00000  1.00000  0.01385
-1.00000  0.00000 -0.00000  0.00000
-0.00000 -1.00000  0.00000  0.96090
 0.00000  0.00000  0.00000  1.00000

_CREO_
[image]
___
## frame=gaze
**q=( 0.00000  0.00000  0.00000  0.00000  0.00000  0.00000)**

_IPOPT_
H=-0.00000  0.00000  1.00000  0.01116
-1.00000  0.00000 -0.00000  0.00000
-0.00000 -1.00000  0.00000  1.01450
 0.00000  0.00000  0.00000  1.00000

_CREO_
[image]

closed 2016-09-23 by fiorisi

pattacini, 2016-09-23:
Thank you @fiorisi !