robotology/cer #71: OpenNI2DeviceServer often does not start

pattacini, 2016-11-25:
We've been often experiencing problems with launching **`OpenNI2DeviceServer`** on `r1-torso`. To our current knowledge, the only workaround is to reboot that machine and 🙏

It would be fairly important to sort this out asap.

cc @vtikha @randaz81 @mbrunettini @maggia80

randaz81, 2017-07-16:
I'm not sure about the name of the module OpenNI2DeviceServer, but this is still happening. In few occasions reboot was not enough and we had to manually disconnect and reconnect the USB cable of the rgbd device from the usb-swich inside R1 torso. Looks like an electrical problem (not enough voltage on the usb bus? or maybe the kernel usb driver is checking the device before it has been powered by the usb hub?) When the problem occurs, a lot of kernel errors are displayed by dmesg.

barbalberto, 2017-07-17:
Just a reminder that `OpenNI2DeviceServer` is deprecated! You should use `depthCamera` device on R1.
The issue does not change because it is a problem between OpenNI2 and the OS, but please use latest software

pattacini, 2018-08-30:
Stale issue that probably won't bother us any longer 🙏

closed 2018-08-30 by pattacini