robotology/cer #74: tripodMotionControl not congruent to simulation model.

ale-git, 2017-03-19:
The `bool TripodSolver::ikin(const Vector &hpr, Vector &lll, int *exit_code)` and `bool TripodSolver::fkin(const Vector &lll, Vector &hpr)` functions implicitly assume that the equivalent joint order is heave->pitch->roll, but in the simulation model the order is heave->roll->pitch instead. They make use of the `dcm2rpy()` and `rpy2dcm()` functions discarding the yaw part, usually small, but they don't give correct values for roll and pitch because they are calculated assuming that pitch comes before roll as in the usual yaw->picth->roll representation.
[https://github.com/robotology/cer/blob/master/cermod/tripod/tripodMotionControl.cpp](url)

pattacini, 2017-03-21:
We agreed to implement equivalent methods `ypr2dcm()` and `dcm2ypr()` in Yarp in order to then fix the problem.

pattacini, 2017-03-21:
@barbalberto the new methods will be called from within `fkin` and `ikin` services, therefore, in principle, no changes are required in `tripodMotionControl`.

pattacini, 2017-03-23:
Upon https://github.com/robotology/yarp/pull/1138 and https://github.com/robotology/cer/commit/840605aeca728a111744a48c26150223282d51d5 it should be fixed by now.

ale-git, 2017-03-23:
Yes, let's close it.

closed 2017-03-23 by ale-git

barbalberto, 2017-03-23:
Great! Thanks