robotology/cer #111: GazeboYarpPlugin stuck from gazebo10

vvasco, 2020-04-27:
Starting from `gazebo10`, `GazeboYarpPlugin` gets stuck here:
https://github.com/robotology/cer/blob/449c4c909b1fbe20169637dfcb7779d0e6fced56/gazeboYarpPlugin/GazeboTripodMotionControl.cpp#L754

The result is that when trying to insert [`cer-sim`](https://github.com/robotology/cer-sim/tree/master/gazebo/cer) into `gazebo` gui, even if the robot is visible and all the ports are opened, the robot is not loaded correctly: 
- when running `yarpmotorgui --robot SIM_CER_ROBOT` it's not possible to control the robot (if trying to move the joint cursor, it gets stuck)
- we cannot receive camera images 

I will have a look to see if I understand what's going on.

cc @pattacini @traversaro @Nicogene

traversaro, 2020-04-27:
It seems that that plugin copied code from an old version of `gazebo_yarp_controlboard` , that send the torque to the gazebo physics engine via a complex pub/sub system, even if both that code and the code that reads the torque are actually executed by the same thread (the main physics thread of Gazebo). 

This fortunatly was fixed by https://github.com/robotology/gazebo-yarp-plugins/pull/324 , and you can find more detail in https://github.com/robotology/gazebo-yarp-plugins/issues/165 . 

Doing the same change in that plugin should be trivial, as as far as I understand, you only need to set the generalized force to the lower level joint.

vvasco, 2020-05-13:
Solved in #115.

closed 2020-05-13 by vvasco