robotology/cer #113: Add support for OpenCV 4

Nicogene, 2020-05-11:
With Ubuntu 20.04 the stable version of opencv became opencv 4.
Under under Ubuntu 20.04 `cer` does not compile.

```
In file included from /home/ngenesio/robotology/cer/modules/faceExpression/main.cpp:26: 
/home/ngenesio/robotology/cer/modules/faceExpression/faceExpressionImage.hpp:5:10: fatal error: opencv/cv.h: No such file or directory 
   5 | #include <opencv/cv.h> 
     |          ^~~~~~~~~~~~~ 
compilation terminated.
In file included from /home/ngenesio/robotology/cer/modules/faceExpression/faceExpressionImage.cpp:3: 
/home/ngenesio/robotology/cer/modules/faceExpression/faceExpressionImage.hpp:5:10: fatal error: opencv/cv.h: No such file or directory 
   5 | #include <opencv/cv.h> 
     |          ^~~~~~~~~~~~~ 
compilation terminated.
make[2]: *** [modules/faceExpression/CMakeFiles/faceExpressionImage.dir/build.make:96: modules/faceExpression/CMakeFiles/faceExpressionImage.dir/main.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [modules/faceExpression/CMakeFiles/faceExpressionImage.dir/build.make:83: modules/faceExpression/CMakeFiles/faceExpressionImage.dir/faceExpressionImage.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1802: modules/faceExpression/CMakeFiles/faceExpressionImage.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/ngenesio/robotology/cer/cermod/faceDisplayServer/faceDisplayServer.cpp:14:10: fatal error: opencv/cv.h: No such file or directory 
  14 | #include <opencv/cv.h> 
     |          ^~~~~~~~~~~~~ 
compilation terminated.
make[2]: *** [cermod/faceDisplayServer/CMakeFiles/faceDisplayServer.dir/build.make:83: cermod/faceDisplayServer/CMakeFiles/faceDisplayServer.dir/faceDisplayServer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```

pattacini, 2020-05-11:
Thanks @Nicogene 
I might have overlooked these bugs as those modules are under conditional compilation if I'm not mistaken.

Nicogene, 2020-05-11:
Fixed by #114

PR #114 (2020-05-11): Add support opencv4
It fixes #113 

Please review code.
closed 2020-05-11 by Nicogene