robotology/cer #126: WHERE IS THE DOCUMENTATION?

FlavioLorenzi, 2022-04-22:
No documentation/tutorial provided and difficulty working with R1 robot!

alecive, 2022-04-22:
Calm down kiddo. You won't get very far by screaming at people.

randaz81, 2022-04-28:
Dear @FlavioLorenzi, this repository is mostly used for internal prototype development.
Since you are writing here, I suppose that you are one of the owners of R1 SN003.
For the usage of that specific robot, especially for the navigation system, please refer to the documentation provided within the joint lab activities.
If you have a general question about the robot software, I recommend opening a discussion in the Q&A section of the community repository: https://github.com/robotology/community
If you have an issue with your specific robot (e.g. hardware fault, system update etc.), please open an issue on the icub-tech support repository: https://github.com/robotology/icub-tech-support

FlavioLorenzi, 2022-04-29:
Hello @randaz81,
thanks for your polite reply. I'll check the links you mentioned.

Dear @alecive , I did not want to disrespect anyone by writing in capital letters nor "scream", I am sorry if you thought this.

alecive, 2022-04-30:
No problem @FlavioLorenzi , just having a bit of fun here :-)

closed 2022-05-06 by FlavioLorenzi